diyless/opentherm-thermostat #1: Question about original thermostat software

geekcoding007, 2024-02-11:
First of all, Keep up the good work!

Is it possible to put on the original file witch contains the original thermostat webpage? (witch also contains the PID settings OT settings and MQTT settings) i made a mistake and erased it accidentally from the ESP board.

diyless, 2024-02-12:
Hi @geekcoding007 
Do you mean the original firmware?

diyless, 2024-02-12:
You can re-flash the original firmware using web flasher https://diyless.com/flasher
Activation code is not required for registered devices.

geekcoding007, 2024-02-12:
Hi @diyless thank you! that was exactly what i needed.

this flasher site comes in very handy if you screw up your firmware 😆 

got the board up and running again 👍.

closed 2024-02-12 by geekcoding007