diyless/opentherm-thermostat #4: [Feature] Expose display state/brightness

ruben-iteng, 2024-12-28:
Expose the current brightness and state of the display + a set point to turn it on/off. 

This way it is possible to use e.g. a motion sensor to turn on the display.

diyless, 2024-12-30:
Nice idea, we can add it to the next update

ruben-iteng, 2025-10-19:
I tested it.
notes:
- `Turn screen off` turns the screen on
- Better rename `Dim screen On` to `Dim screen`
- it would be nice to have a switch (instead of button) for screen On/Off. 1 entity to control instead of 2
- maybe also expose dim and screen off time setting via mqtt. Now the setting need to be set via the display

great work!

diyless, 2025-10-20:
> Turn screen off turns the screen on

Unable to reproduce it. Can you try adding thermostat to HA from scratch?

> Better rename Dim screen On to Dim screen

Will do

> it would be nice to have a switch (instead of button) for screen On/Off. 1 entity to control instead of 2

for now - it is done via buttons because the screen will go dim and then off after the timeouts. If we make it switch - need to update switch state to be disabled when screen of. And what to do if screen is dim?

ruben-iteng, 2025-10-23:
Ah yes true about the switch. I will try to do a clean install to fix the off issue

diyless, 2025-10-26:
@ruben-iteng 
Fixed:
- Turn screen off turns the screen on
- Better rename Dim screen On to Dim screen
- dim / off buttons will be available only if corresponding timeouts are greater than zero
WIP:
- maybe also expose dim and screen off time setting via mqtt. Now the setting need to be set via the display
will drop a message here once test version is ready.

diyless, 2026-09-19:
We have published a `3.0.0` with all the states described above exposed to HA.

closed 2026-09-19 by diyless