diyless/opentherm-thermostat #5: buttons unresponsive after a long time

Helmond, 2025-01-03:
when after the night i try to manualy adjust the temeperature the buttons do not respond. when i change temperature with home assistant, the buttons  do respond.

diyless, 2025-01-03:
Hi @Helmond
Do you mean the physical buttons on the thermostat2?

Helmond, 2025-01-03:
yes, thermostat 2 with sw 2.1

JandeGrootDuracom, 2025-01-04:
Same here with Thermostat2, introduced with 2.0.1 firmware. Previous 1.x versions didn't have this problem.

diyless, 2025-01-09:
@Helmond @JandeGrootDuracom can you show your current configuration?
if you want keeping it private - can send screenshots to our email, we just need to see what is configured and used. We don't need exact physical IP address neither of thermostat itself or the mqtt broker you can blur them if you'd like. Just a statement like _mqtt mosquitto within HA_ or _emqx cluster is used_ etc

Helmond, 2025-01-09:
@diyless 

I have no schedule,
control goes via HA with core-mosquitto mqtt broker.
let me know if you want to know anything else 
[image]
[image]
[image]
[image]

JandeGrootDuracom, 2025-02-15:
Using MQTT with HomeAssistant. Telemetry enabled, Schedule enabled, screen dimming (screen off) enabled. Sensor is the BTLE sensor shipped with thermostat2.

 I was cycling through the screens with top left button and suddenly all buttons stopped working. 1 minute later the buttons worked again.

Thermostat works fine, follows schedule, can be controlled with HA or HTTP and buttons sometimes start working again after a while without rebooting.

This bug also persists with 2.2 firmware.

Helmond, 2025-11-23:
This issue is still persistent in the current software version, it seems the software hangs somewhere. At the moment the webserver also does not respond while the buttons are frozen

diyless, 2026-09-19:
@Helmond we have updated the firmware, can you test newer versions?