diyless/opentherm-thermostat #8: sensor error on 2.1.0 (cannot find BLE any more)

unlucio, 2025-01-20:
I updated my thermostat from 1.10.1 to 2.1.0 and now when I do a BLE scan to find the thermometer, nothing shows up.
I tried
-resetting the thermometer with other GRD and RST pins
- pull the battery, wait a bit, and put it back in again
- moving the thermometer closer to the thermostat

I went back to 1.10.1: it fixed the problem but it stopped talking to the MQTT server, so I tried 2.1.0 again, with the same BLE problem.
I tried to go back to 1.10.1 again and now the diyless.com seems down, so I'm a bit stuck 😅.
I'd try configuring an MQTT sensor for the thermostat, using one of the other thermometers connected to HA and publishing on MQTT, but sits down = doc down 😆

Not my best luck, considering it's January :(

So in summary:
- 2.1.0: no BLW thermometer
- both 2.1.0 and 1.10.1: no MQTT connection

Also, I just found out from this repo that there might be 3 different HW revisions and I've no idea which one I have, I think I got mine back in Jan 2023

diyless, 2025-01-21:
> there might be 3 different HW revisions

We have 3 models of our thermostats, first is just a small board, second has a small display with buttons and third has large square touch display.

> no MQTT connection

it might be that after updating your HomeAssistant has old device config leftovers, since we had a lot of changes in 2.1 regarding the topics, notifying etc. Try removing thermostat completely from MQTT integration and then power on your thermostat again.

Site might be down for a minute or two since hosting does not provide 100% uptime.

> no BLE thermometer

is your thermometer flashed with PVVX?

adrian-sovaila, 2025-09-13:
Same here, from 2.0.1 upwards it can't see any BLE device during scan. Original thermometer.
I even flashed it with your web flasher just to make sure there weren't any leftovers.

unlucio, 2026-02-02:
[image]

[image]

[image]

[image]

idk, I think I did the same, but nothing, if I use a FW > 1.10.1 the sensor is gone

I can't even find a doc on this, is there a tutorial or at least a clear explanation of what is supposed to hapen/ the device expects?

closed 2026-02-02 by unlucio

unlucio, 2026-02-02:
[image]
Now it stopped working with the 1.10.1, too, even after resetting the thermometer to its factory FW, the device also keeps dropping off the network, and the only way to get it back seems to be to power cycle it, and freezing does not exactly help troubleshooting this