diyless/opentherm-thermostat #10: DHW setpoint change does not work after 2.2.0 upgrade

bligeti, 2025-01-25:
Hi,

After reading in the comments that the 2.2.0 is available, I did an OTA updgrade.
On the 2.2.0 the DHW setpoint change is not working for me. The setpoint is set to 80°C - prevously it was on 47°C - and when I tries to change it either via the display or via the web interface or even via MQTT it jumps back to 80°C immedately. In OT messages can see MSGID 56 only wth 80.00, so for me it seems it does not even try to change it.

BR,
Balint

diyless, 2025-01-25:
Which version did you have previously? Do you use Home Assistant?

diyless, 2025-01-25:
You can use update to pull the fixed 2.2, the version remains the same.

bligeti, 2025-01-25:
This was a very quick fix! With the new 2.2.0 it is working fine. Thanks a lot!

closed 2025-01-25 by bligeti

diyless, 2025-01-25:
Sadly, the bug was faster) let us know if any other issues happen