diyless/opentherm-thermostat #11: [FEATURE] Integration of Smart Autotune Thermostat (SAT)

v4l3r10, 2025-02-06:
I would like to propose the integration of support for the [Smart Autotune Thermostat (SAT)](https://github.com/Alexwijn/SAT) into this project. SAT is a custom component for Home Assistant that works with many compatible gateways for boilers. 

I believe that incorporating SAT support would enrich the functionality of this project and provide users with more advanced control over their heating systems.

tshcherban, 2025-03-01:
That might be interesting one. We will investigate what it is and how it works and can be integrated with our thermostat

diyless, 2025-03-01:
That might be interesting one. We will investigate what it is and how it works and can be integrated with our thermostat