diyless/opentherm-thermostat #12: [Enhance] Add End Time for Heating Periods in Schedule

v4l3r10, 2025-02-06:
Currently, the thermostat's scheduling feature allows setting only a start time for heating periods, without specifying an end time. 
This limits the ability to define precise heating intervals and prevents automatic shutdown of the boiler after a desired period. Implementing an option to set both start and end times for heating periods would enable users to define specific time frames during which the heating system is active, allowing for automatic deactivation of the boiler at the end of each scheduled period.

iqlogic, 2025-02-08:
Of course you can set the end time. You just have to pull down the desired temperature. See the attached example:
[image]

v4l3r10, 2025-02-08:
End Time = Off Thermostat . 
In your configuration ur only go down with set point.
Is useful to have something to a complete shutdown.

iqlogic, 2025-02-08:
Start time = Start of heating
End time = End of heating

If set temperature is lower than room temperature the heating is off. I had several thermostats and every single one of them worked this way.

diyless, 2025-02-08:
@iqlogic would it be ok to add a possibility setting `off` in the schedule? Then your end time is just a next interval with `off` as a setpoint.

iqlogic, 2025-02-08:
@diyless It is no problem to me at all. I'm just saying that it has same effect as set temperature lower than room temperature.

v4l3r10, 2025-02-08:
Setting it to OFF is different from simply lowering the temperature because when you lower the temperature, the PI factor continues to operate and make corrections, whereas turning it OFF completely disables the control.

diyless, 2025-02-08:
Ok, we will add that

iqlogic, 2025-02-09:
@v4l3r10 Many thanks for explanation. I didn't think about that this way. I just thought that if I set the temperature to some lower value (temperature in my apartment never drops below 20°C) the heating will be off and thermostat won't operate at all.