diyless/opentherm-thermostat #13: Thermostat 3 shuts off randomly

nenadkrsticnet, 2025-02-09:
Hello, 

With heating curve settings, with hystereses of +1C and lower -0.1C, room SetPoint 21.5, one would assume that the thermostat would send off signal only when room temperature is 22.5C, but in my case it is random, sometime it will turn off at 21.6 sometimes at 21.9, can't find the pattern. There were occasions when it was turn off with outside temperature of +10C, but now it is turning off also at +2C outside. And even when it turns off at 21.9, it doesn't wait for lower hystereses and starts at 21.4, it starts at 21.7...

Based on HA data, I can see that CHEnable form T3 is OFF, but control setpoint is not set to 0, it stays to for example 34C so there seems to be heat demand, but the heat pump is not producing heat as the CHEnable is off. 

What can be the issue?

Now I changed the settings to have lower hysteresis 0, and keeping upper to +1. Also Adjusted a bit heating curve just to try to keep it running at higher outside temperatures. 

[image]

nenadkrsticnet, 2025-02-09:
As a test switched to internal sensor, once reached to 21.8, thermostat off, eliminated BLE drops.

nenadkrsticnet, 2025-02-21:
It is starting to be annoying, looks like as the outside temps are rising the hysteresis is on its own, even with configuration of upper 0.5 and lower 0
10+ C outside, thermostat shuts off at 0.2 difference from set temp. 

@diyless is this something wrong with the configuration, or the calculation is messed up with the T3?

[image]

[image]

[image]

Thank you in advance

diyless, 2025-10-31:
@nenadkrsticnet can you try out the latest update? (update to 2.2.2 and then you will be able to use 2.2.3.b2, beta) we have some fixes related to the heating algorithms used.

diyless, 2025-11-17:
@nenadkrsticnet working on it

diyless, 2025-11-17:
Version 2.2.3.b3 is ready to download

diyless, 2025-11-17:
In fact it supports somewhat more than 14x7 points, but we haven't tested how much so far.
For a (significantly) larger amount we will need to revise schedule endpoint to accept a plain CSV instead of JSON to reduce requests sizes and storage footprint.
How much points are you going to use?

nenadkrsticnet, 2025-11-17:
The maximum I was able to add was 12, as if you add more the t3 restarts, this is more or less testing to have more setpoints in order to keep the heatpump to run longer. 

I would say 15-20 should be more then enough.

diyless, 2026-09-19:
@nenadkrsticnet we have fixed a few related bugs, all of them were around handling large JSON payloads (MQTT, schedule) caused thermostat to hang. For now (3.0.0) it should work. Feel free to reopen or open a new issue if you still see the problem.

closed 2026-09-19 by diyless