diyless/opentherm-thermostat #14: [BUG] Flame State/Level not shown real status

v4l3r10, 2025-02-09:
I am experiencing an anomaly in the thermostat firmware, where the Flame Level value never exceeds 60 and often drops immediately to 0 after reaching this threshold, even though the flame remains on and the water temperature continues to rise. This suggests that the displayed value does not accurately reflect the actual flame status, potentially causing issues with monitoring and combustion regulation.

[image]

v4l3r10, 2025-02-17:
@diyless   
The behavior of the **Flame Level** and **Flame State** is indeed correct as described. However, after further analysis of the boiler's operation, I have found a discrepancy in the **Flame Level** value. Specifically, there is a **1% error** between the boiler's actual power level and what is reported by the thermostat.  

For example:  
- When the boiler is at **80% power**, the thermostat reports **79%**.  
- When the boiler is at **1% power**, the thermostat reports **0%**.  

This discrepancy should be addressed to ensure precise readings. Let me know if I can provide additional data or logs to help refine this.  

Thanks!

vlakant, 2025-02-23:
Hello,

I recently got your DIYLESS OpenTherm Thermostat and integrated it into Home Assistant using MQTT. My boiler is a BAXI DuoTec Compact 1.24. However, since I connected the OT thermostat, I have an issue when heating is not needed. The boiler keeps cycling – it ignites the flame for a few seconds at a flame level of 48% every three minutes.

When there is a heating request from OT, everything works normally. I have gone through and tried various settings on both the boiler and OT, but I can't figure it out.

Do you have any idea what could be causing this issue?

Thank you very much! 

[image]

v4l3r10, 2025-02-24:
@vlakant 

Your BAXI boiler has a minimum power of 3.5 kW, so the frequent ignitions you're experiencing are likely caused by a CH SetPoint that is too low compared to the boiler's minimum power.

To better analyze the situation, could you provide more details about your system? For example:

What type of heating system do you have (underfloor heating, radiators, fan coils)?
What speed is your circulation pump set to?
Could you generate a more detailed graph with additional parameters such as Flame Status, CH Set Point, CH Temperature, etc.?
I’ve encountered a similar issue when the CH Set Point is at 35°C: the boiler, upon ignition, quickly raises the water temperature above 45°C (due to the high minimum power), causing it to shut off immediately and resulting in frequent cycling.

To mitigate this behavior, I adjusted the P18 setting in my boiler’s service menu (which is also available on your BAXI). This setting allows you to configure a minimum waiting time between ignitions, ranging from 10 seconds to 10 minutes. I set it to 10 minutes to prevent constant cycling and improve system stability.

So, the issue is usually due to the boiler’s minimum operating power being too high compared to the requested temperature SetPoint. If your system allows, you might try increasing the CH SetPoint or adjusting the P18 setting to reduce the number of ignitions.

Let me know if you can gather more data so we can better understand how to optimize your boiler’s behavior!

v4l3r10, 2025-02-24:
@vlakant

When I talk about an issue related to minimum power, I'm not referring to a boiler configuration that you can directly modify, but rather to a physical limitation of the system.

Your boiler has a minimum power of **3.5 kW** (as indicated in the table I shared earlier). This means that even at its lowest level, it injects at least **3.5 kW** of thermal energy into the heating circuit. The problem arises when your system requires less energy to maintain the desired temperature.

For example:
- If only **1.2 kW** is needed to maintain the ambient and water temperature,
- But the boiler, even at its minimum, **delivers 3.5 kW**,
- The water temperature will inevitably rise beyond the **CH SetPoint**, causing the flame to shut off and resulting in *short cycling* (frequent on and off cycles).

Modern condensing boilers have significantly improved in terms of modulation capability. In the past, most boilers had a **1:6 modulation ratio**, meaning the minimum power output was one-sixth of the maximum power. More recently, this ratio has improved to **1:8**, and now some high-efficiency boilers feature **1:10 or even 1:12** modulation.

This means that for a **24 kW** boiler:
- With a **1:6 ratio**, the minimum power would be **4 kW**.
- With a **1:10 ratio**, the minimum power would drop to **2.4 kW**.
- With a **1:12 ratio**, it would go down further to **2 kW**.

The lower the minimum modulation power, the better the boiler can adapt to low heating demands, reducing short cycling and improving efficiency. Unfortunately, many older boilers, including those with a **1:6 ratio**, still struggle with maintaining low heat output efficiently, leading to frequent on-off cycles.

One possible way to reduce this behavior could be the introduction of a **hysteresis** parameter in the thermostat. This parameter would define a temperature threshold within which the boiler remains on or off without reacting immediately to every minor fluctuation.

For example, if the hysteresis were set to **2°C**, and the CH SetPoint were **50°C**:
- The boiler would turn on only if the temperature dropped below **48°C**.
- It would remain on until it reaches **52°C**, preventing continuous on and off cycles.

I am not sure if this parameter can be configured via OpenTherm, as it depends on the firmware support of the thermostat and the boiler. For example, on my Ferroli boiler, the hysteresis value is fixed at **5°C** and cannot be changed.

Currently, in the thermostat settings, there doesn't seem to be a dedicated hysteresis parameter, but it could be useful to add it in the future to better manage *short cycling*.

I have been reflecting on possible solutions to this issue for some time. I hope @diyless has some ideas on the matter :)