diyless/opentherm-thermostat #18: [BUG] Dyless weather api not working

v4l3r10, 2025-03-03:
Hi, I noticed that the Weather API function via diyless (weatherapi.com) is not working anymore. Until about 10 days ago, it was showing the outside temperature correctly, but now it just says "Enabled, unavailable" instead. I haven't changed any settings, and it used to work fine before. I'm attaching a screenshot for reference. Let me know if there's anything I can do to fix it. Thanks!

[image]

diyless, 2025-03-03:
We have some troubles with the site hosting, resolving it in the progress

diyless, 2025-10-24:
Weather service was moved to another hosting. Feel free to reopen or open a new issue if you see this problem again.

closed 2025-10-24 by diyless