diyless/opentherm-thermostat #23: [FEATURE REQUEST] RTC handling/showing

MarianRychtecky, 2025-08-31:
Hi all,
I would like to see the current time the thermostat is at. I can't display the current time in 2.2.0. Additionally, I would appreciate an option for NTP settings and/or DHCP Option 042 (RFC 1769) if not supported (unable to check).

Thank you

MarianRychtecky, 2025-10-28:
Hi all, I'm unable to check now, but I think @velzend is correct, as my device is slightly offset by -14 seconds according to the jobs in HA. I assume the RTC is set after the initial flash and keeps it running internally. This is the reason I am requesting the DHCP option. 42 or manual NTP settings.

velzend, 2025-10-30:
This actually made me curious, an offset of -14s... Does the thermostat have a RTC (e.g. DS3231 or DS1307) installed?

diyless, 2025-10-31:
> This actually made me curious, an offset of -14s... Does the thermostat have a RTC (e.g. DS3231 or DS1307) installed?

No, we use ESP32-S3's built-in RTC, which is synced to the following servers:
```
const char *ntpServer1 = "pool.ntp.org";
const char *ntpServer2 = "time.nistv.gov";
const char *ntpServer3 = "time.google.com";
```
using [SNTP Time Synchronization from Espressif](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/system_time.html#sntp-time-synchronization).

> However, I am unable to check, as I cannot find the time anywhere in the WEB interface menu.

You can configure the timezone, and thermostat will try pulling the time via SNTP.

Do we need adding the possibility to set the clock manually and disable internet sync completely?

MarianRychtecky, 2025-10-31:
Hi @diyless, thank you for your comment and support.

My own POV

- To see time in the Web interface 
- Hardcoded SNTP settings are ok with me
- Nice to have would be to see SNTP status in the Web interface (sync status, delta, etc...)
- Option to add own NTP server for isolated networks
- DST settings could be tricky for some users outside the EU
- Manual time set is nice to have, but not really needed for me

diyless, 2026-04-30:
@MarianRychtecky we have added ntp retriveal from dhcp (`sntp_servermode_dhcp(1);`). Will let you know once we publish a test firmware with it.

diyless, 2026-05-31:
@MarianRychtecky try `2.3.0.alpha`

diyless, 2026-09-17:
DHCP Option 042 (RFC 1769) is active, should be working on versions >= 2.3.0.alpha2. Feel free to reopen the issue if needed.

closed 2026-09-17 by diyless